AresValley/Artemis #6: Artemis crash when download database

cn0xroot, 2019-07-16:
artemis looks nice, but it don't work on osx, It crash when download database

os : 10.14.5
artemis Version 3.0.0

```
Process:               artemis [21534]
Path:                  /Applications/Artemis.app/Contents/MacOS/artemis
Identifier:            Artemis
Version:               0.0.0 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           artemis [21534]
User ID:               503

Date/Time:             2019-07-16 17:18:11.520 +0800
OS Version:            Mac OS X 10.14.5 (18F132)
Report Version:        12
Bridge OS Version:     3.5 (16P5125)
Anonymous UUID:        67502330-0894-1289-F4CC-0316B9715F7F

Sleep/Wake UUID:       0404262B-0933-4EF4-A22F-957EC2EE0CE2

Time Awake Since Boot: 480000 seconds
Time Since Wake:       64000 seconds

System Integrity Protection: enabled

Crashed Thread:        14  DownloadThread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff6cd6822a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff6cd6876c mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff40d05bee __CFRunLoopServiceMachPort + 328
3   com.apple.CoreFoundation      	0x00007fff40d0515c __CFRunLoopRun + 1612
4   com.apple.CoreFoundation      	0x00007fff40d048be CFRunLoopRunSpecific + 455
5   com.apple.HIToolbox           	0x00007fff3fff096b RunCurrentEventLoopInMode + 292
6   com.apple.HIToolbox           	0x00007fff3fff06a5 ReceiveNextEventCommon + 603
7   com.apple.HIToolbox           	0x00007fff3fff0436 _BlockUntilNextEventMatchingListInModeWithFilter + 64
8   com.apple.AppKit              	0x00007fff3e38a987 _DPSNextEvent + 965
9   com.apple.AppKit              	0x00007fff3e38971f -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1361
10  com.apple.AppKit              	0x00007fff3e38383c -[NSApplication run] + 699
11  libqcocoa.dylib               	0x000000011fdafccb 0x11fd7c000 + 212171
12  QtCore                        	0x000000011db0750f QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 431
13  QtCore                        	0x000000011db0c452 QCoreApplication::exec() + 130
14  QtWidgets.so                  	0x000000011cb097d2 meth_QApplication_exec_(_object*, _object*) + 82
15  libpython3.7m.dylib           	0x000000010f0a3213 _PyMethodDef_RawFastCallKeywords + 259
16  libpython3.7m.dylib           	0x000000010f0a34bd _PyCFunction_FastCallKeywords + 45
17  libpython3.7m.dylib           	0x000000010f1ab16b call_function + 603
18  libpython3.7m.dylib           	0x000000010f1a2365 _PyEval_EvalFrameDefault + 7893
19  libpython3.7m.dylib           	0x000000010f19fff9 _PyEval_EvalCodeWithName + 3305
20  libpython3.7m.dylib           	0x000000010f19f300 PyEval_EvalCode + 48
21  Artemis                       	0x000000010eab6fa9 0x10eab5000 + 8105
22  Artemis                       	0x000000010eab755a 0x10eab5000 + 9562
23  Artemis                       	0x000000010eab5cb4 0x10eab5000 + 3252

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff6cd6b86a __psynch_cvwait + 10
1   libsystem_pthread.dylib       	0x00007fff6ce2a56e _pthread_cond_wait + 722
2   libopenblasp-r0.3.7.dev.dylib 	0x00000001106aac0b blas_thread_server + 619
3   libsystem_pthread.dylib       	0x00007fff6ce272eb _pthread_body + 126
4   libsystem_pthread.dylib       	0x00007fff6ce2a249 _pthread_start + 66
5   libsystem_pthread.dylib       	0x00007fff6ce2640d thread_start + 13
[... 496 lines cut ...]
    0x7fff6ce1a000 -     0x7fff6ce23fef  libsystem_platform.dylib (177.250.1) <16949870-9152-3B0D-9EF0-40FAF84B0F1E> /usr/lib/system/libsystem_platform.dylib
    0x7fff6ce24000 -     0x7fff6ce2eff7  libsystem_pthread.dylib (330.250.2) <2D5C08FF-484F-3D59-9132-CE1DCB3F76D7> /usr/lib/system/libsystem_pthread.dylib
    0x7fff6ce2f000 -     0x7fff6ce32ff7  libsystem_sandbox.dylib (851.260.2) <9CACC5F5-3859-3A1F-BCC6-96DDD4A556A8> /usr/lib/system/libsystem_sandbox.dylib
    0x7fff6ce33000 -     0x7fff6ce35ff3  libsystem_secinit.dylib (30.260.2) <88F3CE61-5FAD-3CDA-AA13-B5B18B3EBD26> /usr/lib/system/libsystem_secinit.dylib
    0x7fff6ce36000 -     0x7fff6ce3dff3  libsystem_symptoms.dylib (820.267.1) <03F1C2DD-0F5A-3D9D-88F6-B26C0F94EB52> /usr/lib/system/libsystem_symptoms.dylib
    0x7fff6ce3e000 -     0x7fff6ce53fff  libsystem_trace.dylib (906.260.1) <100013AE-0443-3CF0-AC17-5D934608B60B> /usr/lib/system/libsystem_trace.dylib
    0x7fff6ce55000 -     0x7fff6ce5affb  libunwind.dylib (35.4) <24A97A67-F017-3CFC-B0D0-6BD0224B1336> /usr/lib/system/libunwind.dylib
    0x7fff6ce5b000 -     0x7fff6ce8afff  libxpc.dylib (1336.261.2) <9A0FFA79-082F-3293-BF49-63976B073B74> /usr/lib/system/libxpc.dylib

External Modification Summary:
  Calls made by other processes targeting this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by this process:
    task_for_pid: 0
    thread_create: 0
    thread_set_state: 0
  Calls made by all processes on this machine:
    task_for_pid: 502434
    thread_create: 0
    thread_set_state: 0
```

can you share database download url? thanks.

cn0xroot, 2019-07-16:
python version on github is work fine.

deanpemberton, 2019-07-17:
Just putting this here to maybe help other people.
The .dmg version was crashing for me so I tried the python version.

I was getting a certificate error when trying to download the data file.
```
ssl.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)
```

Turns out this was a known issue which the following link fixed

https://stackoverflow.com/questions/42098126/mac-osx-python-ssl-sslerror-ssl-certificate-verify-failed-certificate-verify

Best of luck

commit e5e80b69 (2019-07-31): Fix #6 Add cacert.pem also for async downloads
closed 2019-07-31 by alessandro90