AresValley/Artemis #16: Impossible to run 3.2.0 version on Arch Linux

dvdesolve, 2020-02-08:
Trying to run it different ways. I'm on the latest kernel, Python version is 3.8.1 and Qt is 5.14.1. All requirements are installed.

---
## Binary from aresvalley.com
When running from terminal I can see splash screen for a moment and then nothing happens. In terminal output I see the following messages:
```
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
Gtk-Message: 21:09:51.427: Failed to load module "canberra-gtk-module"

(Artemis:82708): GdkPixbuf-WARNING **: 21:09:51.439: Cannot open pixbuf loader module file '/usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders.cache': No such file or directory

This likely means that your installation is broken.
Try running the command
  gdk-pixbuf-query-loaders > /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders.cache
to make things work again for the time being.
Fontconfig warning: "/etc/fonts/fonts.conf", line 5: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/fonts.conf", line 6: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/fonts.conf", line 6: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/fonts.conf", line 6: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/fonts.conf", line 7: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/fonts.conf", line 7: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/fonts.conf", line 9: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/30-metric-aliases.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/30-metric-aliases.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/30-metric-aliases.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/30-metric-aliases.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/40-nonlatin.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/40-nonlatin.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/40-nonlatin.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/40-nonlatin.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/45-generic.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/45-generic.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/45-generic.conf", line 6: invalid attribute 'xmlns:its'
Fontconfig error: "/etc/fonts/conf.d/45-generic.conf", line 6: invalid attribute 'version'
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 8: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 4: unknown element "its:rules"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 5: unknown element "its:translateRule"
Fontconfig error: "/etc/fonts/conf.d/45-latin.conf", line 5: invalid attribute 'translate'
Fontconfig error: "/etc/fonts/conf.d/45-latin.conf", line 5: invalid attribute 'selector'
Fontconfig error: "/etc/fonts/conf.d/45-latin.conf", line 6: invalid attribute 'xmlns:its'
[... 90 lines cut ...]
Traceback (most recent call last):
  File "site-packages/PyInstaller/loader/rthooks/pyi_rth_pkgres.py", line 13, in <module>
  File "/usr/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 489, in exec_module
    exec(bytecode, module.__dict__)
  File "site-packages/pkg_resources/__init__.py", line 84, in <module>
ModuleNotFoundError: No module named 'packaging.requirements'
[71677] Failed to execute script pyi_rth_pkgres
```

I can workaround it by changing `hiddenimports=[],` line to the `hiddenimports=['packaging.requirements'],` in `Artemis.spec`. However I can't still run it after rebuild because of the following:
```
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
Traceback (most recent call last):
  File "artemis.py", line 769, in <module>
  File "artemis.py", line 215, in __init__
  File "themesmanager.py", line 290, in start
    self._detect_themes()
  File "themesmanager.py", line 143, in _detect_themes
    ag = QActionGroup(self._owner, exclusive=True)
TypeError: 'exclusive' is an unknown keyword argument
[82600] Failed to execute script artemis
```

alessandro90, 2020-02-29:
Hi @dvdesolve ,

I don't know exactly the reason for the first error, looks like there is a problem with the possibility to change the font added in the new release.

To compile from source: actually I still have to try Python ver. 3.8 to run Artemis. I'm not sure all the dependencies are satisfied. Apart from that, note that the PyQt version you are using is not the one specified in the requirements.txt file. In particular note that in your PyQt version the 'exclusive' parameter is in fact not available: https://github.com/metabrainz/picard/pull/1424

I suggest to build a Python 3.7.0 environment and install the dependencies as per the requirements.txt file and try again.

dvdesolve, 2020-02-29:
I can't figure out why it complains about `canberra-gtk-module` - I already have `libcanberra` installed:
```
[viktor@desolve-nettop ~]$ pacman -Ql libcanberra | grep canberra-gtk-module
libcanberra /etc/X11/xinit/xinitrc.d/40-libcanberra-gtk-module.sh
libcanberra /usr/lib/gnome-settings-daemon-3.0/gtk-modules/canberra-gtk-module.desktop
libcanberra /usr/lib/gtk-2.0/modules/libcanberra-gtk-module.so
libcanberra /usr/lib/gtk-3.0/modules/libcanberra-gtk-module.so
```
Regarding GdkPixbuf error: seems like path `/usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders.cache` is hardcoded in binary somehow so it tries to look in non-existent place.
Don't know why there are errors about Fontconfig but how can I figure out the cause of the segfault?

Thank you for Python 3.7 suggestion. I'll give it a try. According to your link it seems like `exclusive` parameter is not needed anymore. Will new release support new Qt?

alessandro90, 2020-03-01:
The cause of the segfault seems to be in the setupUi function, which is a PyQt function. We tested the binary versions on Linux Mint and Fedora, so I think it is possible that the compiled version could not function properly in all linux distros.

I would try to just run the script `python artemis.py` first (given you are in a proper python environment with all the requirements satisfied). If that works, I'd try to compile from the .spec file.

About the Qt 5.14 support: chances are that the `exclusive` parameter may be removed also in the currently supported version. I will try to see if everything works the same and eventually remove it.

dvdesolve, 2020-03-01:
Ok, thank you

benreese0, 2020-03-30:
Not sure if it helps, but I modified the code to remove the 'exclusive' parameter

Artemis/src/themesmanager.py line 143
old: `ag = QActionGroup(self._owner, exclusive=True)`
new: `ag = QActionGroup(self._owner)`

and so far it seems to be working fine. (Fairly new Artemis user, so I'm not sure if I'm missing something).

commit cfd302d3 (2020-04-18): Close #16 Remove 'exclusive' parameter from a PyQt function. Also update readme.md
closed 2020-04-18 by alessandro90