AresValley/Artemis #17: .dmg cannot be mounted on Mac OS (before Sierra)

CD3XT3RW4RD, 2020-04-07:
When downloading .dmg installer, it cannot be mounted on certain systems, with error "no mountable file system". I assume problem is that .dmg was created in APFS filesystem, which is not supported OS versions older than Sierra. Can you pls generate .dmg not in APFS?

AresValley, 2020-05-07:
From the version 3.2.1 we have an experimental legacy release for macOS: https://drive.google.com/open?id=1GEw4c8LS6nEZOPoSGW7nNCsFUz614mzm
Let us know if it works!

CD3XT3RW4RD, 2020-05-07:
Quickly tested just the mount and it went fine even on my medieval El Capitan :)
Will test functionality in the evening.

AresValley, 2020-05-08:
Excellent, you are welcome! I'll add the legacy version to the main download page

closed 2020-05-08 by AresValley