AresValley/Artemis #19: Auto ID functions

WheezyE, 2020-04-09:
In addition to auto-correlation, Artemis might benefit from being able to switch between other methods of signal ID - kind of [like PEiD used to do](https://www.aldeid.com/wiki/PEiD#Menu) for using different scanning methods to find what kind of packers were on executables).  A user might select between auto-correlation ID, or [fast Fourier transform ID](http://jcrueda.com/?p=916), or [deep learning](https://www.rtl-sdr.com/shazam-style-automatic-signal-identification-via-the-sigidwiki-database/#comment-167727).  Maybe a meta-ID feature could be created that runs all three ID functions on the same audio capture, then picks the output that had the highest confidence.

I think you guys may already be working on this (I saw MarcoDT's insightful comments on the above-referenced RTL-SDR article & JC Rueda artile), but I wanted to document this possible enhancement in the official Git to have a thread for discussion.

AresValley, 2024-06-09:
@WheezyE Hey Eric, I hope you're doing great! It's kind of a weird way to contact you, but I couldn't find any other way...I updated Artemis and, with that, also the main site [https://www.aresvalley.com](https://www.aresvalley.com)
Since you did a huge amount of work for porting Artemis to Raspberry, I find it right to include you on the website as well (in the acknowledgments section). I also linked Winelink as a featured/highlighted project with other friends' projects. I just wanted to let you know, and if you wish to change something or have something removed, I will edit it right away based on your input.
The project email is: aresvalley.info (at) gmail.com if you wish to write me in private

WheezyE, 2024-06-10:
Hey MarcoDT, It's great to hear from you! I love the new features in Artemis 4.0 and the new website! I am truly honored to be included on your site. I've learned a lot about bash scripting since 2020 and am excited to start working on compiling Artemis 4.0 for my Pi's soon. I'll try to work on it as soon as possible and I'll definitely keep you updated via email. I hope you're doing great too :) Feel free to reach out any time
73, de KI7POL

AresValley, 2024-06-10:
Awesome! Well, this is the minimum for me since you did a lot of tests and work behind Raspberry compatibility. Unfortunately, I added you a little late because I was hoping to update the site much sooner. I still have to try on my Raspberry Pi 2B v1.1 because I was waiting to prepare some minor hotfix just after the release. Anyway, sure! I look forward to hearing from you, and I'm also curious about how the new Artemis is running on ARM devices :) Since the number of third-party libraries has dropped considerably (PySide6, requests, and packaging), hopefully, compatibility will be much easier.
Thanks again for your time Eric!