AresValley/Artemis #28: Crash when trying to play audio

MightyPork, 2021-12-22:
artemis3 installed from AUR on arch linux

```
pygame 2.1.0 (SDL 2.0.18, Python 3.10.1)
Hello from the pygame community. https://www.pygame.org/contribute.html
22/12/2021 11:28:11 AM::ERROR::updatescontroller::_check_updater_version::Unable to query the updater
Traceback (most recent call last):
  File "audio_player.py", line 146, in _play_audio
    self._set_max_progress_bar()
  File "audio_player.py", line 122, in _set_max_progress_bar
    self._audio_progress.setMaximum(
TypeError: setMaximum(self, int): argument 1 has unexpected type 'float'
/usr/bin/artemis3: line 4:  3456 Aborted                 (core dumped) /opt/artemis3/Artemis "$@"
```

MightyPork, 2022-02-18:
easy fix, cast to int

```
    def _set_max_progress_bar(self):
        """Set the maximum value of the progress bar."""
        self._audio_progress.setMaximum(
            int(mixer.Sound(self._audio_file).get_length() * 1000)
        )
```

cmpsvictor, 2022-08-20:
happens on Manjaro as well, same fix worked

AresValley, 2022-09-30:
Fixed in #34

closed 2022-09-30 by AresValley