AresValley/Artemis #30: Crash on X11 Arch

Frigyes06, 2022-02-13:
I initially had problems with even starting Artemis on my Arch with wayland, but it did not work, so I switched it to X11. now I get the following errors:
```
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
/bin/sh: /tmp/_MEITJ7LW1/libncursesw.so.6: no version information available (required by /usr/lib/libreadline.so.8)
/usr/lib/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
Failed to load module: /usr/lib/gio/modules/libgvfsdbus.so

(Artemis:4187): GdkPixbuf-WARNING **: 22:01:11.237: Cannot open pixbuf loader module file '/usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders.cache': No such file or directory

This likely means that your installation is broken.
Try running the command
  gdk-pixbuf-query-loaders > /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders.cache
to make things work again for the time being.
Gtk-Message: 22:01:11.250: Failed to load module "canberra-gtk-module"
Gtk-Message: 22:01:11.250: Failed to load module "canberra-gtk-module"
Fontconfig warning: "/etc/fonts/fonts.conf", line 5: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/48-spacing.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 4: unknown element "description"
Fontconfig warning: FcPattern object weight does not accept value [40 210)
Fatal Python error: (pygame parachute) Segmentation Fault

Current thread 0x00007f32f9a82740 (most recent call first):
  File "<string>", line 37 in setupUi
  File "artemis.py", line 85 in __init__
  File "artemis.py", line 775 in <module>
Aborted (core dumped)
```

Frigyes06, 2022-02-13:
fixed this error:
```
(Artemis:4187): GdkPixbuf-WARNING **: 22:01:11.237: Cannot open pixbuf loader module file '/usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders.cache': No such file or directory

This likely means that your installation is broken.
```
by creating /x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/ then making a symlink:
```
ln -s /usr/lib/gdk-pixbuf-2.0/2.10.0/loaders.cache /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders.cache
```

pinkavaj, 2022-09-27:
Works for me on Arch using Wayland.

rupertdev, 2023-05-05:
similar crash on Arch X11 i3wm: 
```
~/Downloads/artemis
❯ ./Artemis
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
/bin/sh: /tmp/_MEIPVXAfC/libncursesw.so.6: no version information available (required by /usr/lib/libreadline.so.8)
Fontconfig warning: "/etc/fonts/fonts.conf", line 5: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-sub-pixel-rgb.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-yes-antialias.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/48-spacing.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 4: unknown element "description"
Fontconfig warning: FcPattern object weight does not accept value [40 210)
Fatal Python error: (pygame parachute) Segmentation Fault

Current thread 0x00007f6beb873740 (most recent call first):
  File "<string>", line 37 in setupUi
  File "artemis.py", line 85 in __init__
  File "artemis.py", line 775 in <module>
Aborted (core dumped)
```

AresValley, 2024-06-01:
This issue is derived from the pygame library. Artemis v4 is rolling out these days and does not rely on the pygame library anymore for multimedia reproduction

closed 2024-06-01 by AresValley