AresValley/Artemis #33: Binary does not work without network

Devian9, 2022-08-19:
Program does not run from binary without network. Database is downloaded. We can run it only with python3 from source code.

AresValley, 2022-09-30:
Need more info about OS, version, ... Not reproducible at the moment with the current version of Artemis

AresValley, 2024-06-01:
> > Need more info about OS, version, ... Not reproducible at the moment with the current version of Artemis
> 
> @AresValley, encountered same issue, here is more information:
> 
> Ubuntu 22.04.4 LTS; 64-bit OS; Artemis Mk. III version 3.2.1
> 
> Here is the console stack trace:
> 
> ```
> pygame 1.9.6
> Hello from the pygame community. https://www.pygame.org/contribute.html
> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
> /usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
> Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
> /usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so: undefined symbol: g_byte_array_steal
> Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so
> /tmp/_MEIF88INv/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /lib/x86_64-linux-gnu/libproxy.so.1)
> Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiolibproxy.so
> /usr/lib/x86_64-linux-gnu/gio/modules/libgiognomeproxy.so: undefined symbol: g_task_set_name
> Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiognomeproxy.so
> /usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so: undefined symbol: g_task_set_name
> Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgioremote-volume-monitor.so
> 
> (Artemis:64818): Gtk-WARNING **: 20:10:17.406: Theme parsing error: gtk.css:1413:23: 'font-feature-settings' is not a valid property name
> 
> (Artemis:64818): Gtk-WARNING **: 20:10:17.408: Theme parsing error: gtk.css:3286:25: 'font-feature-settings' is not a valid property name
> 
> (Artemis:64818): Gtk-WARNING **: 20:10:17.408: Theme parsing error: gtk.css:3748:23: 'font-feature-settings' is not a valid property name
> Fontconfig warning: "/etc/fonts/fonts.conf", line 5: unknown element "its:rules"
> Fontconfig warning: "/etc/fonts/fonts.conf", line 6: unknown element "its:translateRule"
> Fontconfig error: "/etc/fonts/fonts.conf", line 6: invalid attribute 'translate'
> Fontconfig error: "/etc/fonts/fonts.conf", line 6: invalid attribute 'selector'
> Fontconfig error: "/etc/fonts/fonts.conf", line 7: invalid attribute 'xmlns:its'
> Fontconfig error: "/etc/fonts/fonts.conf", line 7: invalid attribute 'version'
> Fontconfig warning: "/etc/fonts/fonts.conf", line 9: unknown element "description"
> Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "its:rules"
> Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: unknown element "its:translateRule"
> Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: invalid attribute 'translate'
> Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 5: invalid attribute 'selector'
> Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 6: invalid attribute 'xmlns:its'
> Fontconfig error: "/etc/fonts/conf.d/10-hinting-slight.conf", line 6: invalid attribute 'version'
> Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 8: unknown element "description"
> Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 4: unknown element "its:rules"
> Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 5: unknown element "its:translateRule"
> Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 5: invalid attribute 'translate'
> Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 5: invalid attribute 'selector'
> Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 6: invalid attribute 'xmlns:its'
> Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 6: invalid attribute 'version'
> Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 8: unknown element "description"
> Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 4: unknown element "its:rules"
> Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 5: unknown element "its:translateRule"
> Fontconfig error: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 5: invalid attribute 'translate'
> Fontconfig error: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 5: invalid attribute 'selector'
> Fontconfig error: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 6: invalid attribute 'xmlns:its'
> Fontconfig error: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 6: invalid attribute 'version'
> Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 8: unknown element "description"
> Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 4: unknown element "its:rules"
> Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 5: unknown element "its:translateRule"
> Fontconfig error: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 5: invalid attribute 'translate'
> Fontconfig error: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 5: invalid attribute 'selector'
> Fontconfig error: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 6: invalid attribute 'xmlns:its'
> Fontconfig error: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 6: invalid attribute 'version'
> Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 8: unknown element "description"
> Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 4: unknown element "its:rules"
[... cut ...]
> urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='aresvalley.com', port=443): Max retries exceeded with url: /Storage/Artemis/Package/latest_versions.json (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7b7f4118a710>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
> Aborted
> ```

Artemis 4 is rolling out these days. The 2nd RC is available here: [https://github.com/AresValley/Artemis/releases](https://github.com/AresValley/Artemis/releases). I'll close for now since the problem should be solved due to the total different framework

closed 2024-06-01 by AresValley