AresValley/Artemis #38: Win 7 install

fellowpacker, 2023-06-04:
Program doesn't install on win 7 without https://github.com/nalexandru/api-ms-win-core-path-HACK modified "api-ms-win-core-path-l1-1-0.dll"  in windows system32 and syswow64. Maybe that should either included, noted or referenced somewhere. ;-)

fellowpacker, 2023-06-04:
Once that is installed it works like a charm under win 7

AresValley, 2023-06-04:
Thanks for the report. Support for Windows 7 has been removed a few months ago: the latest python that can be used on Windows 7 is python 3.8. The site was updated some time ago but the readme on GitHub unfortunately was not.

Artemis 4 is under (very) active development, so within a few weeks (I hope), the page will be completely updated!

closed 2024-02-20 by AresValley