AresValley/Artemis #42: won't run on wayland

ToontjeM, 2024-02-11:
```
$ ./Artemis 
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

(Artemis:5207): GLib-GIO-ERROR **: 19:05:01.525: Settings schema 'org.gnome.settings-daemon.plugins.xsettings' does not contain a key named 'antialiasing'
Trace/breakpoint trap (core dumped)
```
Ubuntu 22.04 on NUC i7.

ToontjeM, 2024-02-11:
Disabling wayland makes it work.

closed 2024-02-11 by ToontjeM

bmn001, 2024-05-21:
@ToontjeM How did you do this?  Did you just downgrade your entire desktop to x11?

With X11 increasingly being phased out on major desktop distros, it seems like Wayland support really should be added.  I'd suggest this bug should be reopened.

AresValley, 2024-05-21:
> @ToontjeM How did you do this? Did you just downgrade your entire desktop to x11?
> 
> With X11 increasingly being phased out on major desktop distros, it seems like Wayland support really should be added. I'd suggest this bug should be reopened.

Artemis 4 will be released in a few weeks or even less. The new software is already on testing and now I'm preparing all the new documentation. Hopefully all the compatibility problems will be resolved soon

AresValley, 2024-06-01:
> @ToontjeM How did you do this? Did you just downgrade your entire desktop to x11?
> 
> With X11 increasingly being phased out on major desktop distros, it seems like Wayland support really should be added. I'd suggest this bug should be reopened.

Artemis 4 - RC2 is available ;)