AresValley/Artemis #43: 4.0 RC Do not launch on Windows 10

dmaltsiniotis, 2024-06-01:
I have downloaded the latest RC, and after installation, a window quickly appears then closes right away. The application does not seem to launch correctly after fresh install. Is there something I can to do provide more details or logs? There is nothing in the Application or System event logs about a crash or issue. There are no log files in the Artemis directory either.

OS: Windows 10 Pro 22H2
RAM: 64 GB
CPU: 18 Core Intel 7980XE

Thank you,

-DM

AresValley, 2024-06-01:
Thanks for the report! Something that may have changed, and I only just noticed, was the release of Nuitka 2.3 (which happened a few hours ago), and all the tests on the machines available to me use the latest version that was available, 2.2.3. I just rebuilt the program with Nuitka 2.3 and it seems to work normally but I need to do more extensive testing. In the meantime, I forced the standard output and error generation in the folder where Artemis is installed. You can find **artemis.out.txt** and **artemis.err.txt**when running the software. The latter should contain something useful in case of a crash.
Just download the updated installer again to give it a try.

dmaltsiniotis, 2024-06-03:
Hi there, thanks for the super quick reply on this. I re-downloaded the RC2 release, ran the installer, and unfortunately still ran into some issues. The behavior is a little different now, the app still crashes on startup but now prompts me to optionally attach a debugger. Also, I looked in the install path (for me C:\Program Files (x86)\Artemis) and I don't see any artemis.out.txt or artemis.err.txt files generated.

0RedCell0, 2024-06-03:
Team Artemis,the "update" executable gets highlighted by VIRUSTOTAL as Malware/Trojan...

AresValley, 2024-06-03:
> Hi there, thanks for the super quick reply on this. I re-downloaded the RC2 release, ran the installer, and unfortunately still ran into some issues. The behavior is a little different now, the app [...]

Thanks for all the details. Now I am pretty sure it has something to do with Nuitka. With Artemis 3 we were usued to package the software with pyinstaller but now has been changed to Nuika for several reasons. However, if it becomes problematic I can revert to the old packaging option. To confirm the cause of the problem you can try to run the code directly from source following the instructions in [documentation](https://aresvalley.github.io/Artemis/run_from_source/)

AresValley, 2024-06-03:
Thanks for letting me know. This is a very well known problem of nuitka and Windows (https://nuitka.net/user-documentation/common-issue-solutions.html#windows-virus-scanners) and it is simply a false positive. Only 1 out of 74 antivirus detects it as a false positive so not bad at all to be honest (https://www.virustotal.com/gui/file/4c194876cac0defb7fc05777fdb175394ccc6f4054fb09f71ed4d2dbfa465274). 

On the other hand, can you give me more details about the executable 'update' and can you share me the link of virustotal because I cannot see any scans done before of the artemis executables nor the installer...

dmaltsiniotis, 2024-06-03:
> > Hi there, thanks for the super quick reply on this. I re-downloaded the RC2 release, ran the installer, and unfortunately still ran into some issues. The behavior is a little different now, the app still crashes on startup but now prompts me to optionally attach a debugger. Also, I looked in the install path (for me C:\Program Files (x86)\Artemis) and I don't see any artemis.out.txt or artemis.err.txt files generated:
> > ![image](https://private-user-images.githubusercontent.com/106233/336248953-887f215c-9049-4e1c-9791-569019039c3a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0NTUxODIsIm5iZiI6MTcxNzQ1NDg4MiwicGF0aCI6Ii8xMDYyMzMvMzM2MjQ4OTUzLTg4N2YyMTVjLTkwNDktNGUxYy05NzkxLTU2OTAxOTAzOWMzYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwM1QyMjQ4MDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNmE0NTc4ZjE4YjRmMTljYTBhOTY2NmIyNzAzM2NmYjQ2NjMzODdmZGEwYzZlNjgxY2M4YThjODhiMjhlYzc5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.g1Vz5bJyCgMQlJR2yHpLhgfQ22UDUcd4S8D85S3wHi8)
> 
> @dmaltsiniotis thanks for all the details. Now I am pretty sure it has something to do with Nuitka. With Artemis 3 we were usued to package the software with pyinstaller but now has been changed to Nuika for several reasons. However, if it becomes problematic I can revert to the old packaging option. To confirm the cause of the problem you can try to run the code directly from source following the instructions in [documentation](https://aresvalley.github.io/Artemis/run_from_source/)

Okay so I tried a few things:

1) Running from source, the app works great with no issues.

2) I used the Windows PowerShell build scripts to build a fresh app.dist with Nutika, and I launched from the app.dist folder, and that also worked (albeit with a slight issue of not being able to download the database.

I'm not sure why the published release version isn't working for me.

AresValley, 2024-06-03:
@dmaltsiniotis Good news! I was able to reproduce the problem, so I labeled this issue as a confirmed bug. The problem is the installation path 'C:\Program Files (x86)\Artemis'. Artemis is not able to start and any _out_ and _err_ files are created in the folder. The `program file` and `program files (x86)` directories seem to show the same behavior but any other installation folders (not system protected, of course) seems to works fine. If you want to temporary solve the problem just install Artemis in the standard `C:\Users\$USER\AppData\Local\Programs` folder or any other one of your choice (desktop, documents, etc)

dmaltsiniotis, 2024-06-03:
> @dmaltsiniotis Good news! I was able to reproduce the problem, so I labeled this issue as a confirmed bug. The problem is the installation path 'C:\Program Files (x86)\Artemis'. Artemis is not able [...]

Sounds good, thank you for helping me work through the analysis!

AresValley, 2024-06-03:
> Okay so I tried a few things:
>
> 1. Running from source, the app works great with no issues.
> 2. I used the Windows PowerShell build scripts to build a fresh app.dist with Nutika, and I launched from [...]

Thanks to you for the quick reply and all the really precious testing you have done! Now I think I have sufficient material to work with in order to solve this issue. One last thing: do you still have issue to download the SigID database from the installed version or it is a common problem of all the version you have (running Artemis from source and from the binary)?

0RedCell0, 2024-06-03:
Aloha!

Stand by brother...more data and screenshots on da way!

🤙🏽
Dominik
WB2CAT

0RedCell0, 2024-06-03:
Absolutely!  As a Ham Radio Operator I deeply respect your work & time!!

AresValley, 2024-06-03:
Thanks, Dominik! I greatly appreciate the help. Artemis underwent a big change from version 3 to 4 so we have the usual problems that we also had during the launch of the previous version

dmaltsiniotis, 2024-06-04:
> Thanks to you for the quick reply and all the really precious testing you have done! Now I think I have sufficient material to work with in order to solve this issue. One last thing: do you still [...]

I was able to download SigID database with no issues when running the app from source as well as after installing to a non-protected directory, all seems well. I suspect when I had issues, that it was related to some python module problems I had during the build process.

[image]

closed 2024-06-04 by AresValley

AresValley, 2024-06-04:
The issue has been solved and was due to the fact that the Program Files and Program Files (x86) folder (as any other protected ones) allow read-only operations with normal privileges. Now Artemis uses a folder structure (OS dependant) that splits read/write ops and read-only operations: in this way you can install Artemis in any folder you wish without elevated privileges. This was a large issue remained unresolved from Artemis Mk.3

Everything has been documented here:
https://aresvalley.github.io/Artemis/run_from_source/#folders-structure

It will be available in Artemis 4.0.0-RC3 very soon. Thanks for the help!

commit 4e7ebcc2 (2024-06-04): Read-only ops are separated in different standard and OS dependent folders from read-write ones (fixed #43), bump Nuitka 2.3