AresValley/Artemis #48: App.bin not launching 4.0.1 on 22.04

alphafox02, 2024-06-10:
On 22.04 I've unzipped the latest release and correctly ran chmod 700 on the app.bin. When trying to run the app.bin nothing happens. I checked ldd against it and it seems like everything is there that it needs. Switching to running 4.0.1 from source using python does work however.

AresValley, 2024-06-10:
Thanks for the report! Can you show me the error log in in the **/tmp** folder? The file is called **artemis.err.log**
I bet is something related to xcb...

AresValley, 2024-06-10:
Run the app.bin and the check the log file. Otherwise should be empty if the application runs from the source code properly

alphafox02, 2024-06-10:
I'll check, but I was just coming back to say that when I compile the application with the script you all provide, they app.bin works. I'll go check the packaged one.

alphafox02, 2024-06-10:
So oddly enough, I just pulled down the release zip again, unzipped it and ran chmod 700 app.bin. It starts without issue. Now I'm wondering if perhaps there's something that got installed when I ran the build script you provide (it pulled a lot of pip packages and I had to install patchelf for it to build. I'll have to try on another fresh 22.04.

AresValley, 2024-06-10:
> I'll check, but I was just coming back to say that when I compile the application with the script you all provide, the app.bin works. I'll go check the packaged one.

Perfect! Yes, this behavior is not new to me and also happens to compile the software with Nuitka 2.3.1 or 2.3.2 on GitHub machines. Using Nuitka 2.3 solves the problem, but if I try compiling with Nuitka 2.3.1/2.3.2 on my fedora, it works fine. If I'm correct, it should be something similar:
```
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
```

AresValley, 2024-06-10:
> So oddly enough, I just pulled down the release zip again, unzipped it and ran chmod 700 app.bin. It starts without issue. Now I'm wondering if perhaps there's something that got installed when I [...]

Cool...i'm investigating...I'm not 100% sure it's a problem with Artemis but more with libraries that the OS has to deafult or not...

alphafox02, 2024-06-10:
After making the utils folder I can see there may be some deeper issues with qt, but it could just be my setup. I'm building from source again on the other pc.

AresValley, 2024-06-10:
okay, this is quite strange...in principle artemis uses the qtquickcontrols2.conf in the config and copy the content in ~/.local/share/AresValley/Artemis/config folder...from your error it seems that the program is not able to locate the config folder inside the main Artemis folder but I just checked and the file is there. Now I'll try to do a fresh install on my fedora, let see if I'm able to reproduce the problem

alphafox02, 2024-06-10:
Okay may have figured out possibly the bigger cause

Host needs libcxb-cursor0 installed. That allows me to now run the app on the other computer. It’s possibly my laptop had this package. Now I should start all over and repeat steps.

AresValley, 2024-06-10:
Okay i was able to reproduce the problem about the qtquickcontrols2.conf in a particular condition so I labeled the issue as a confirmed bug. I'll check if I'm able to locate where it comes from. I'll look forward to your tests. In the meantime thanks for the help!

closed 2024-06-10 by AresValley

AresValley, 2024-06-10:
If you wish to test on your machine:

1. delete the `.local/share/AresValley` folder to prepare a fresh environment
2. download the updated source code and compile the binary as you did previously
3. chmod 700 app.bin and then launch the application. If everything goes well, the program should start and the `.local/share/AresValley` folder should be created

alphafox02, 2024-06-10:
I’ll go try again. Did you also catch this was needed?

libcxb-cursor0

maybe unique to me, but without it I was getting a seg fault and could see Artemis was complaining it needed it. Sudo apt installed fixed that right up.

AresValley, 2024-06-11:
Nope, that's one is a different story I guess. This problem is due to a missing system library as reported here

https://bugreports.qt.io/browse/PYSIDE-2306
https://bugreports.qt.io/browse/QTBUG-112185
https://bugreports.qt.io/browse/PYSIDE-2301
https://bugreports.qt.io/browse/QTBUG-67373

in the first link, it seems to affect an Ubuntu distribution 22.04 LTS, like the one you mentioned in the first post. Pretty sure is not a bug related to Artemis but more to a missing dependency of PySide6. However, xcb-cursor0 or libxcb-cursor0 should be a present dependency in 6.5(dev) and later, so this is kind of strange. Anyway, could be a regression...nothing I can do on my side but update the documentation regarding this peculiar situation

AresValley, 2024-06-12:
I completely agree. The documentation has just been updated (https://aresvalley.github.io/Artemis/installation/#linux). Many are the dependencies needed to make X11 work properly (https://doc.qt.io/qt-6/linux-requirements.html), but most of the time, they are provided by the system, and the only one that still causes problems is the cursor one, i.e., libxcb-cursor0. The documentation will be updated accordingly if we encounter any other issues with dependencies. Since at least 15 libxcb libraries are required, it does make sense to do something like `sudo apt install libxcb*`, but I want to avoid bloating systems for no reason. Also libx11-dev and libx11-xcb-dev can be missing but this scenario is way less frequent. Thanks again @alphafox02 and @maboegit for the help! Thanks also for the news about TechMinds! ì

commit 436c54b7 (2024-06-10): Fixed #48 an error occurig on Linux where configuration file path are not properly resolved during startup with the binary version of the program