AresValley/Artemis #49: Cannot download database

MiloKerrigan, 2024-06-12:
Hello, I get the following error:

No SigID database detected... Do you want to download it now --yes

Something went wrong! An error occured during the process. Details: The credentials were not recognized / Invalid argument --OK

Any ideas? Is there a way to download the database manually? Using Win10 and ver 4.0.3 Thanks.

AresValley, 2024-06-12:
This is pretty strange because that error message takes place when you have to authenticate yourself when accessing the given URL. This does not makes sense because the DB is here: https://github.com/AresValley/Artemis-DB/releases/download/v61/v61.tar and, as you can see, no authentication is required. The only thing I'm thinking is: do you have some restriction on your network that does not allow you to access the GitHub repo here (https://github.com/AresValley/Artemis-DB) or GitHub in general?

MiloKerrigan, 2024-06-12:
Well that's a mystery! I downloaded the v61.tar with no problems at all in firefox, then told Artemis to import database and it works fine. Thanks for your prompt help and a fantastic program.

closed 2024-06-12 by MiloKerrigan

AresValley, 2024-06-12:
Awesome! Thanks a lot to you @MiloKerrigan and feel free to reopen the issue if the problem persists!