AresValley/Artemis #52: I can't find basic signals like adsb

poetajuan, 2024-06-15:
Very good application, congratulations.
I don't know why the basic frequency of the ADSB is not set, there is a similar one but it does not correspond in frequency and name. The correct one is Automatic Dependent Surveillance-Transmission (ADS-B) 978MHz - 1,090MHz.

AresValley, 2024-06-15:
Thanks @poetajuan! I can't find the problem you described. This is what I see by selecting ADS-B
[image]
and this is exactly corresponding to sigid wiki (https://www.sigidwiki.com/wiki/Automatic_Dependent_Surveillance-Broadcast_(ADS-B)). Maybe I misunderstood your question. Can you give me more details?

AresValley, 2024-06-16:
Closing for lack of feedback and because the alleged problem, with the given information, is not reproducible.
Feel free to open the issue again with more details

closed 2024-06-16 by AresValley