AresValley/Artemis #54: Cannot auto download SigID database at first start: TLS error

mrkeuz, 2024-07-19:
### Problem

After first start Artemis proposes to download SigID database, but fail with "TLS error".

### Solved
   via manually load database from:
   https://github.com/AresValley/Artemis-DB/releases/download/v65/v65.tar  

### Environment
- Ubuntu 20.04 x64 
- Artemis 4.0.5 (version downloaded from https://www.aresvalley.com/)

### Steps to reproduce:
- Clean installation **WITHOUT** database cause after database been imported dialog does not appear.
- Then dialogs:
[image]
[image]

AresValley, 2024-07-20:
Thanks for the excellent report. I'm now investigating if it is an issue I can reproduce in your same starting conditions

AresValley, 2024-07-20:
This is the dump of the error and the problem is due to an incompatible openSSL version. The problem is well documented.

```
qt.multimedia.ffmpeg.libsymbolsresolver: Couldn't load OpenSsl library
qt.multimedia.ffmpeg.libsymbolsresolver: Couldn't load VAAPI library
qt.tlsbackend.ossl: Incompatible version of OpenSSL (built with OpenSSL >= 3.x, runtime version is < 3.x)
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslKey
qt.network.ssl: Active TLS backend does not support key creation
qt.network.ssl: The backend "cert-only" does not support QSslSocket
qt.network.ssl: The backend named "cert-only" does not support TLS
qt.network.ssl: QSslSocket::connectToHostEncrypted: TLS initialization failed
QNetworkReplyImplPrivate::error: Internal problem, this method must only be called once.
```
A solution will be available on the next release

AresValley, 2024-07-20:
This problem, in case of Ubuntu, is specific only for the 20.x versions since they are shipped with the openSSL version 1. The later versions (22, 24) includes the openSSL version 3 so the problem is not present.

I was able to solve the problem following this instruction (https://orcacore.com/install-openssl-3-ubuntu-20-04/) on how to download and compile the openSSL 3 for Ubuntu 20.04