AresValley/Artemis #57: Add a manual sync option for database update

Saharagnome, 2024-08-30:
There exists a feature to auto load the latest version of the SigID for the database downloaded from SigWiki. Since I don't want to use the cycles to call home every time it loads I leave it unchecked. This means that when I do want it to update I have to manually download the new one or re-enable auto update, close, then reopen the program. Would it be possible to add a manual-update/sync option to grab the latest data only when I need it?

AresValley, 2024-08-31:
I think I did not fully understand the problem. The checkbox in the preferences only loads the latest database available on the local (hence offline) machine from where the program is launched. Regardless of whether the database is loaded automatically or not, Artemis compares the last database available offline with the last one available online. **The database does not have to be loaded for this feature to be available.** If you want to check manually, you can do so from the menu: Help->Check for updates. The program will look for updates to the software and database.
Did I misread the issue you described?

AresValley, 2024-10-20:
Closing for lack of feedback and because the alleged problem, with the given information, is not reproducible.
Feel free to open the issue again with more details

closed 2024-10-20 by AresValley