AresValley/Artemis #63: flatpak package

parkan, 2024-12-03:
Hello, I use an atomic desktop Linux distribution (Fedora CoreOS/Silverblue derivative) where GUI apps are typically packaged as flatpaks. The Artemis tool runs reasonably well, but installation would be much simpler and cleaner as a flatpak. Would you be open to me creating a community-contributed flatpak on Flathub?

AresValley, 2024-12-05:
Hi Arkadiy, I find it basically a very good idea. There has never been much user demand for a container package, but I can see its potential usefulness for Linux or even arm architectures, at this point. 
Artemis is an open source software so user contributions are more than welcome!

My only priority is to keep the flatpack on flathub up-to-date and in line with GitHub releases. The version on flathub should never be different from the latest available on GitHub only because no user has generated the flatpack and/or uploaded it to flathub. Therefore, I could make some suggestions on how to proceed:

1) If you have already had a chance to test/generate the flatpack you can make a pull request of the manifest in this repo.

2) Add to the git-action that generates the build for Linux, the generation of the flatpack as well. By doing this, in fact, when the program is compiled by nuitka, the flatpack is also generated and is ready to be uploaded directly to flathub.

Do you agree or do you have another idea?

parkan, 2025-01-19:
@AresValley my apologies, somehow GitHub seems to have eaten the notification for this. I think (2) is the right idea (as relying on downstream maintainer availability is clearly not ideal as evidenced by the turnaround time on this from me 😅)

Will update with flatpak build results shortly.