AresValley/Artemis #66: Idea: Android App

emptycall, 2024-12-23:
Hi Guys,

Absolutely great work on Artemis - brilliant project.
An android version (that can work offline) would be so so useful - id happily donate and help drum up support if this is possible.

Many thanks

bloudman648, 2025-01-11:
please do it 😎

dsalomon, 2026-04-09:
FYI, I tried to install & run under UserLAnd on my Samsung Fold 5.  I was able to download and unpack the current Linux .zip file, then chmod to make app.bin executable.  However, when I try to run it (./app.bin or even sudo ./app.bin) I get a not found error.  I'm sure this has something to do with trying to run in the UserLAnd virtual environment, but I don't know enough about that package to debug this.  It was worth a shot.  Any suggestion welcome.