AresValley/Artemis #67: Build on MacOS Fail

JimByers, 2025-01-10:
M4  Mac  / OS 15.2

Will run from source:  Need  to manually load DataBase most of the time  no other problems so far. (4 days)

Build from source: Fatal  error  related to  PySide6 and Nuika
         https://bugreports.qt.io/browse/PYSIDE-2956?jql=project%20%3D%20PYSIDE%20AND%20component%20%3D%20Tooling

I have no skill or experience in  programming.  Very much a cut and paste and see what happens.

Jim
VE7KAJ

JimByers, 2025-02-06:
Build on Mac OS 15.3  M4Pro

The issues so far
Text size is too small in the Signal descrpition field 
Audio files in .ogg will not play (Mac OS issue Apple dropped support for .ogg)
Database needs to be installed at every startup.

Jim
VE7KAJ

editrice, 2026-07-04:
Hi,
just built it fine today on macOS26 arm.
All I had to change from the install instruction was pipx (and not py3) install briefcase

editrice, 2026-07-04:
@AresValley Yes using pipx is the recommended way to install py3 packages/scripts if they aren't on homebrew on macOS.
For the font yes it's a bit small but still ok, only in the description field, rest is fine, might be high dpi display?
I haven't had a problem with playing sounds tho I didn't play all of them (and I don't have issues playing .ogg in general)
The database is to load at every restart, don't know if it's the same behavior on other platforms.
The .app bundle with the db downloaded is 1.35GB, not sure if that's excepted, on the heavy side